ghi-electronics/duelink-website #193: CAN bus

gus-ghielec, 2025-04-19:
I am only reading zeros back. I set mode to 0x80 (config mode) but then reading status is always 0. Status should change to 0x80.

```
statled(200,200,0)
#cs 15
# https://github.com/autowp/arduino-mcp2515/tree/master

fn rdReg(r)
    dwrite(15,0)
    SpiWr(3)#INSTRUCTION_READ)
    SpiWr(r)
    r=SpiWr(0)
    dwrite(15,1)
    return r
fend

fn modReg(r,m, d)
    dwrite(15,0)
    SpiWr(5)#INSTRUCTION_BITMOD);
    SpiWr(r)
    SpiWr(m)
    SpiWr(d)
    dwrite(15,1)
fend

fn setMode(m)
    modReg(0x0f,0xE0,m)#MCP_CANCTRL, CANCTRL_REQOP, m);

    while 1
        if rdReg(0x0E)= m# MCP_CANSTAT
            return
        end
        println("wait ", m)
        wait(1000)

    wend

fend

fn setReg(r,v)
    dwrite(15,0)
    SpiWr(0x02)#INSTRUCTION_WRITE)
    SpiWr(r)
    SpiWr(v)
    dwrite(15,1)
fend

#define MCP_8MHz_500kBPS_CFG1 (0x00)
#define MCP_8MHz_500kBPS_CFG2 (0x90)
#define MCP_8MHz_500kBPS_CFG3 (0x82)

fn setBitrate()
    setMode(0x80)#CANCTRL_REQOP_CONFIG);
    setReg(0x2A,0)# MCP_CNF1
    setReg(0x29,0x90)# MCP_CNF2
    setReg(0x28,0x82)#MCP_CNF3
fend

fn sndMsg(i,b1,l)#const TXBn txbn, const struct can_frame *frame)
    if l > 8
        return
    end

    #const struct TXBn_REGS *txbuf = &TXB[txbn];

    dim b2[13];

    #bool ext = (frame->can_id & CAN_EFF_FLAG);
    #bool rtr = (frame->can_id & CAN_RTR_FLAG);
    #uint32_t id = (frame->can_id & (ext ? CAN_EFF_MASK : CAN_SFF_MASK));

    # we will leave id to 0 for now = prepareId(data, ext, id);

    #data[MCP_DLC] = rtr ? (frame->can_dlc | RTR_MASK) : frame->can_dlc;
    b2[4] = l
    
    #memcpy(&data[MCP_DATA], frame->data, frame->can_dlc);
    for i in range(l)
        b2[5+i]=b1[i]
    next
    
    #setRegisters(txbuf->SIDH, data, 5 + l);
    setRegs(0x31,b2,5+l)
    
    #modifyRegister(txbuf->CTRL, TXB_TXREQ, TXB_TXREQ);
    modReg(0x30,0x08,0x08)
    
    #uint8_t ctrl = readRegister(txbuf->CTRL);
    r=rdReg(0x08)
    
    print(r)
fend

########
SpiCfg(0, 8000)
dwrite(15,1)

SpiWr(0xC0)#INSTRUCTION_RESET);
wait(100)
setBitrate()
setMode(0)#CANCTRL_REQOP_NORMAL);

dsfasdfsad
dim b1[8]=[0x11,0x22]

while 1
    sndMsg(0,b1,2)
    wait(1000)
wend
```

Palomino34, 2025-04-22:
this one for sending only

```
fn rdReg(r)
    dwrite(15,0)
 
    SpiWr(3)#INSTRUCTION_READ)
    SpiWr(r)
    r=SpiWr(0)
 
    dwrite(15,1)
    return r
fend
 
fn modReg(r,m, d)
    dwrite(15,0)
 
    SpiWr(5) #INSTRUCTION_BITMOD);
    SpiWr(r)
    SpiWr(m)
    SpiWr(d)
 
    dwrite(15,1)
fend
 
fn setMode(m)
    modReg(0x0f,0xE0,m)#MCP_CANCTRL, CANCTRL_REQOP, m);
 
    while 1
        if rdReg(0x0E)= m# MCP_CANSTAT
            println("match ", m)
            return
        end
        println("wait ", m)
        wait(1000)
 
    wend
 
fend
 
fn setReg(r,v)
    dwrite(15,0)
    SpiWr(0x02)#INSTRUCTION_WRITE)
    SpiWr(r)
    SpiWr(v)
    dwrite(15,1)
fend
 
fn setRegs(r, b2, n)
    dwrite(15,0)
 
    SpiWr(0x02)
    SpiWr(r)
 
    for i in range(0,n)
        SpiWr(b2[i])
    next
 
    dwrite(15,1)
fend
 
#define MCP_8MHz_500kBPS_CFG1 (0x00)
#define MCP_8MHz_500kBPS_CFG2 (0x90)
#define MCP_8MHz_500kBPS_CFG3 (0x82)
 
fn setBitrate()
    setMode(0x80)#CANCTRL_REQOP_CONFIG);
    setReg(0x2A,0)# MCP_CNF1
    setReg(0x29,0x90)# MCP_CNF2
    setReg(0x28,0x82)#MCP_CNF3
fend
 
 
fn sndMsg(b1,l)#const TXBn txbn, const struct can_frame *frame)
    if l > 8
        return
    end
 
    # r=rdReg(0x30) # TQD check TXBnCTRL TX
 
    #println("=>>>>Check TX reg: ", r)    
 
    #if (r & (1<<4))
    #    println("There is tx pending: ", r)    
    #end    
 
    #const struct TXBn_REGS *txbuf = &TXB[txbn];
 
    dim b2[13]
 
    #bool ext = (frame->can_id & CAN_EFF_FLAG);
    #bool rtr = (frame->can_id & CAN_RTR_FLAG);
    #uint32_t id = (frame->can_id & (ext ? CAN_EFF_MASK : CAN_SFF_MASK));
 
    # we will leave id to 0 for now = prepareId(data, ext, id);
 
    #data[MCP_DLC] = rtr ? (frame->can_dlc | RTR_MASK) : frame->can_dlc;
    b2[4] = l
 
 
    #memcpy(&data[MCP_DATA], frame->data, frame->can_dlc);
    for i in range(l)
        b2[5+i]=b1[i]
    next
 
    #setRegisters(txbuf->SIDH, data, 5 + l);
    setRegs(0x31,b2,5+l)
 
    #modifyRegister(txbuf->CTRL, TXB_TXREQ, TXB_TXREQ);
    modReg(0x30,0x08,0x08)
 
    #uint8_t ctrl = readRegister(txbuf->CTRL);
    r=rdReg(0x30)
 
    if ((r & (0x40 | 0x20 | 0x10 )) != 0) 
        println("send has error: ", r)
    else
        println("no error found: ", r)
    end
 
fend
 
fn Init() 
 
    dwrite(15,1)
 
    dwrite(15,0)
    SpiWr(0xC0)#INSTRUCTION_RESET);
    dwrite(15,1)
 
    wait(10)
 
    println("Set bitrate ")
    setBitrate()
    
    println("Set normal")
    setMode(0)#CANCTRL_REQOP_NORMAL);
fend 
 
fn getStat()
    dwrite(15,0)
    SpiWr(0xA0)#INSTRUCTION_READ_STATUS );
    r=SpiWr(0)
    dwrite(15,1)
 
    return r
fend
 
 
####################################
 
_i = 0 
 
 
fn DoTest() 
    SpiCfg(0, 8000)
    Init() 
    println("1 Status: ", getStat())
    dim b1[8]=[0x11,0x22,0x33,0x44,0x55,0x66,0x77,0x88]
 
    println("Sending cnt: ", _i) 
 
    Wait(1000)
    sndMsg(b1,8)
 
    _i = _i + 1
 
    Wait(1000)
   println("2 Status: ", getStat())
 
fend 
```

Palomino34, 2025-05-02:
This is for both send and receiving:
There is an issue send twice with same message. We may read datasheet to see if clear the sending request

```
fn rdReg(r)
    dwrite(15,0)
 
    SpiWr(3)#INSTRUCTION_READ)
    SpiWr(r)
    r=SpiWr(0)
 
    dwrite(15,1)
    return r
fend

fn rdRegs(r, b1, o,l)
    dwrite(15,0)
 
    SpiWr(3)#INSTRUCTION_READ)
    SpiWr(r)
    
    for i = o to (l + o)
        b1[i]=SpiWr(0)
    next
 
    dwrite(15,1)
fend
 
fn modReg(r,m, d)
    dwrite(15,0)
 
    SpiWr(5) #INSTRUCTION_BITMOD);
    SpiWr(r)
    SpiWr(m)
    SpiWr(d)
 
    dwrite(15,1)
fend
 
fn setMode(m)
    modReg(0x0f,0xE0,m)#MCP_CANCTRL, CANCTRL_REQOP, m);
 
    while 1
        if rdReg(0x0E)= m# MCP_CANSTAT
            #println("match ", m)
            return
        end
        #println("wait ", m)
        wait(1000)
 
    wend
 
fend
 
fn setReg(r,v)
    dwrite(15,0)
    SpiWr(0x02)#INSTRUCTION_WRITE)
    SpiWr(r)
    SpiWr(v)
    dwrite(15,1)
fend
 
fn setRegs(r, b2, n)
    dwrite(15,0)
 
    SpiWr(0x02)
    SpiWr(r)
 
    for i in range(0,n)
        SpiWr(b2[i])
    next
 
    dwrite(15,1)
fend
 
#define MCP_8MHz_500kBPS_CFG1 (0x00)
#define MCP_8MHz_500kBPS_CFG2 (0x90)
#define MCP_8MHz_500kBPS_CFG3 (0x82)
 
fn setBitrate()
    setMode(0x80)#CANCTRL_REQOP_CONFIG);
    setReg(0x2A,0)# MCP_CNF1
    setReg(0x29,0x90)# MCP_CNF2
    setReg(0x28,0x82)#MCP_CNF3
fend
 
 
fn sndMsg(b1,l)#const TXBn txbn, const struct can_frame *frame)
    if l > 8
        return
    end
 
    # r=rdReg(0x30) # TQD check TXBnCTRL TX
 
    #println("=>>>>Check TX reg: ", r)    
 
    #if (r & (1<<4))
    #    println("There is tx pending: ", r)    
    #end    
 
    #const struct TXBn_REGS *txbuf = &TXB[txbn];
 
    dim b2[13]
 
    #bool ext = (frame->can_id & CAN_EFF_FLAG);
    #bool rtr = (frame->can_id & CAN_RTR_FLAG);
    #uint32_t id = (frame->can_id & (ext ? CAN_EFF_MASK : CAN_SFF_MASK));
 
    # we will leave id to 0 for now = prepareId(data, ext, id);
 
    #data[MCP_DLC] = rtr ? (frame->can_dlc | RTR_MASK) : frame->can_dlc;
    b2[4] = l
 
 
    #memcpy(&data[MCP_DATA], frame->data, frame->can_dlc);
    for i in range(l)
        b2[5+i]=b1[i]
    next
 
    #setRegisters(txbuf->SIDH, data, 5 + l);
    setRegs(0x31,b2,5+l)
 
    #modifyRegister(txbuf->CTRL, TXB_TXREQ, TXB_TXREQ);
    modReg(0x30,0x08,0x08)
 
    #uint8_t ctrl = readRegister(txbuf->CTRL);
    r=rdReg(0x30)
 
    if ((r & (0x40 | 0x20 | 0x10 )) != 0) 
        println("send has error: ", r)
    else
        println("no error found: ", r)
    end
 
fend

fn ReadMsg(b1)
    # b1
    # byte 0,1,2,3: id
    # byte 4, length
    # byte 5: data
    
    
    #readRegisters(rxb->SIDH, tbufdata, 5)
    rdRegs(0x61, b1, 0, 5) # MCP_RXB0SIDH =  0x61, should scan for 6 rx fifo. Now just trying fifo 0
    
    

    
    #uint32_t id = (tbufdata[MCP_SIDH]<<3) + (tbufdata[MCP_SIDL]>>5)
    d = (b1[0]<<3) + (b1[1]>>5)
    
    if ( (b1[1] & 0x8) =  0x8 ) 
        d = (d<<2) + (b1[0]<<b1[1] & 0x03)
        d = (d<<8) + b1[0]<<b1[2]
        d = (d<<8) + b1[0]<<b1[3]
        d |= 0x80000000
    end
    
    
    l = b1[4] & 0x0F # length
    
    if (l = 0)
        return 0
    end    
    
    
    c = rdReg(0x60) # MCP_RXB0CTRL
    
    if (c & 0x08)
        d = d | 0x40000000 # RTR
    end
    
    
    rdRegs(0x66, b1, 5, l) # MCP_RXB0DATA 0x66
    
    modReg(0x2C, 0x1, 0) #MCP_CANINTF  , CANINTF_RX0IF, 
    
   
    return l
fend

fn Init() 
 
    dwrite(15,1)
 
    dwrite(15,0)
    SpiWr(0xC0)#INSTRUCTION_RESET);
    dwrite(15,1)
 
    wait(10)
 
    println("Set bitrate ")
    setBitrate()
    
    println("Set normal")
    setMode(0)#CANCTRL_REQOP_NORMAL);
fend 
 
fn getStat()
    dwrite(15,0)
    SpiWr(0xA0)#INSTRUCTION_READ_STATUS );
    r=SpiWr(0)
    dwrite(15,1)
 
    return r
fend
 
 
####################################
 
_i = 20
 
 
fn DoTest() 
    SpiCfg(0, 8000)
    Init() 

    _i = 20
    c = 0
    while (_i > 0 )
        s = getStat()
        
        println("Status: ", s)
        
        Wait(1000)
        #dim b1[8]=[0x11,0x22,0x33,0x44,0x55,0x66,0x77,0x88]
        
        #sndMsg(b1,8)
        
        
        
        if s & (1<<0)
            dim b1[13]
            l = ReadMsg(b1)
            
            
            
            if ReadMsg(b1)
                for i = 0 to 13
                    println("data ", b1[i])
                next 
                
        
                dim b1[8]=[0x11,0x22,0x33,0x44,0x55,0x66,0x77,0x88]
                sndMsg(b1,8)
                
                println("rec l: ", l, ", c = ", c )
                c = c + 1
            end
        end
        
        
        _i = _i -1
    wend
    
 

 
fend 
```