ghi-electronics/duelink-website #197: NRF24

gus-ghielec, 2025-04-24:
The datasheet is very nice and clear https://cdn.sparkfun.com/datasheets/Wireless/Nordic/nRF24L01_Product_Specification_v2_0.pdf
The chip has "pipes" that carry an address and then a payload of up to 32 bytes can be sent.
The chip reset values are good, like pipe 0 and 1 are already enabled and have an address. Just put payload and send!
There is dynamic payload and "Enhanced ShockBurst" mode that I did not read about. No need to support any of this. We do nto need to even support pipes! Just a very simple driver that sends and receives up to 32 bytes from a specific address at 1Mbpp.

This is what an API might look like for example
```
#send
RF24Init()
while 1
b1=[1,2,3,4,5]
RF24Send(0x12345678,b1,5)
wait(1000
wend
```

```
#receive
RF24Init()
RF24Listen(0x12345678)
while 1
if RF24Available()
len = RF24Read(b1)
dump(b1)
end
wend
```

Here is a starting point

```
statled(200,200,5)

# https://cdn.sparkfun.com/datasheets/Wireless/Nordic/nRF24L01_Product_Specification_v2_0.pdf

#19 CE
#16 CSN
#15 IRQ

#commands
#0x00 + (reg 0x1F) read reg (5bytes)
#0x20 + (reg 0x1F) write reg (5 bytes)
#0x61 read rx payload (32 bytes)
#0xa0 write tx payload (32 bytes)
#0xE1 flush TX
#0xE2 flush RX
#0xe3 reuse tx payload
#0x50 activate???
#0x60 read rx payload width
#0xa8 + (0x07 PPP) in rx mode to write with ACK???
#0xB0 (32 bytes) txmode without auto ack
#0xFF NOP

fn rdReg(r)
    dwrite(16,0)#CS
    SpiWr(0x00|r)
    r=SpiWr(0xff)#nop
    dwrite(16,1)#CS
    return r
fend

fn wrReg(r,v)
    dwrite(16,0)#CS
    s = SpiWr(0x20|r)
    SpiWr(v)
    dwrite(16,1)#CS
    return s#status
fend

fn status()
    dwrite(16,0)#CS
    s=SpiWr(0xff)
    dwrite(16,1)#CS
    return s
fend

fn rdpl()
    dwrite(16,0)#CS
    s=SpiWr(0x61)#R_RX_PAYLOAD
    for i in range(32)
    b1[i] = SpiWr(0xFF)
    next
    dwrite(16,1)#CS
fend

fn wrpl()
    dwrite(16,0)#CS
    SpiWr(0xa0)#W_TX_PAYLOAD
    
    #for i in range(32)
        SpiWr(0x55)#one bytes for now
    #next
    dwrite(16,1)#CS
fend

SpiCfg(0, 8000)
dwrite(19,0)#enable
dwrite(16,1)#CS

wrreg(6,0)#RF_SETUP 1Mbpp
#wrreg(0,2)#CONFIG power up TX
#wrreg(0,3)#CONFIG power up RX
wrreg(7,(1<<6)|(1<<5)|(1<<4))#STATUS clear RX_DR TX_DS MAX_RT

#wrreg(5,2???)#RF_CH
SpiWr(0xE1)#FLUSH_TX
SpiWr(0xE2)#FLUSH_RX

#tx def add 0xE7E7E7E7E7
#pipe0 def addr 0xE7E7E7E7E7
#pipe1 def addr 0xC2C2C2C2C2
#pipe2 = pipe1 +1 ...etc and these are only LSB
#wrregSSS(address) # RX_ADDR_P1
#wrreg(3,3)#SETUP_AW defult is 5(0x03)
#wrreg(2,(1<<1))#EN_RXADDR enable pipe 1 (0 and 1 are enabeled on power up)

#start writing ###############
wrreg(0,2) # CONFIG = PWR_UP PRIM_TX
#tx add and rx pipe 0 are def to 0xE7E7E7E7E7
wrreg(0x11,1)#RX_PW_P0 why do we write to this if TX?! I don't think this is needed
wait(1)
wrpl()
dwrite(19,1)#enable
wait(1)
dwrite(19,0)#enable

_s=rdreg(8)#OBSERVE_TX
println(_s)
_s=wrreg(7,(1<<6)|(1<<5)|(1<<4))#status clear and read in one shot RX_DR TX_DS MAX_RT
println(_s)

#start listening ###############
wrreg(0,3) # CONFIG = PWR_UP PRIM_RX
wrreg(0x11,1)#RX_PW_P0 bytes used, must match on both ends
dwrite(19,1)#enable
wait(1)

println("waiting...")

while 1
_s = status()

if _s & (1<<6) #RX_DR
    println("rx pipe: ", (_s>>1)7)
    statled(100,100,0)
    #rdpl()
    
    wrreg(7,(1<<6))#clear status
end

wait(1000)
wend
```

Palomino34, 2025-05-01:
These for RX, TX on NRF24 min

```
 
Alias(ce=19,csn=16,intPin=15)
 
fn WriteReg(r, v)
    dwrite(csn, 0)
 
    if r < 0x20
        spiwr(0x20 | (r & 0x1F))
        spiwr(v)
    else
        spiwr(r)
 
        if ((r != 0xE1) && (r != 0xE2) && (r != 0xE3) && (r != 0xFF))
            spiwr(v)
        end
    end
 
    dwrite(csn, 1)
fend
 
fn ReadReg(r)
    dwrite(csn, 0)
 
    spiwr(r & 0x1F)
 
    v =  spiwr(0xFF)
 
    dwrite(csn, 1)
    
    return v
fend
 
fn WriteRegs(r, b1, c)
    dwrite(csn, 0)
 
    spiwr(r)
    for i = 0 to c
        spiwr(b1[i])
    next
 
    dwrite(csn, 1)
fend
 
fn ReadRegs(r, b1, c)
    dwrite(csn, 0)
 
    spiwr(r)
    for i = 0 to c
        b1[i] = spiwr(0xFF)
    next
 
    dwrite(csn, 1)
fend
 
#fn ClrIrq() 
#    r = ReadReg(0x07)
#    r = r | 0x70
#    WriteReg(0x07, r)
#fend
 
fn setAdd(p, b1)
    if (p = 0  || p = 1 || p = 6)
        w = ReadReg(0x3) + 1
 
        i = 0
        i = i + w
 
[... 295 lines cut ...]
            if ReadPl(b1, b2, 0) > -1
                for i = 0 to b2[0]
                    println(" ", b1[i])
                next
                
                if f = 0
                    statled(50, 50, 0)
                    f = 1
                end
            end
            
        else
            println(" No thing comming yet")
            #statled(0, 1, 0)    
        end
        
        
        Wait(1000)
        
        println("Wait....")
    wend
 
 
fend
```

Palomino34, 2025-06-26:
Update use pin irq instead of pull register

Note: Need to clear interrupt register after package received

```
Alias(ce=19,csn=16,intPin=1)
 
fn WriteReg(r, v)
    dwrite(csn, 0)
 
    if r < 0x20
        spiwr(0x20 | (r & 0x1F))
        spiwr(v)
    else
        spiwr(r)
 
        if ((r != 0xE1) && (r != 0xE2) && (r != 0xE3) && (r != 0xFF))
            spiwr(v)
        end
    end
 
    dwrite(csn, 1)
fend
 
fn ReadReg(r)
    dwrite(csn, 0)
 
    spiwr(r & 0x1F)
 
    v =  spiwr(0xFF)
 
    dwrite(csn, 1)
 
    return v
fend
 
fn WriteRegs(r, b1, c)
    dwrite(csn, 0)
 
    spiwr(r)
    for i = 0 to c
        spiwr(b1[i])
    next
 
    dwrite(csn, 1)
fend
 
fn ReadRegs(r, b1, c)
    dwrite(csn, 0)
 
    spiwr(r)
    for i = 0 to c
        b1[i] = spiwr(0xFF)
    next
 
    dwrite(csn, 1)
fend
 
#fn ClrIrq() 
#    r = ReadReg(0x07)
#    r = r | 0x70
#    WriteReg(0x07, r)
#fend
 
fn setAdd(p, b1)
    if (p = 0  || p = 1 || p = 6)
        w = ReadReg(0x3) + 1
 
        i = 0
        i = i + w
[... 333 lines cut ...]
    Init()
    SetMode(0, b1, 32, 1)
 
    # send payload
    dim b1[32]
 
    for i = 0 to 32
     b1[i] = i
    next
 
    while (1)
 
        if Transmit(b1,32)
            println("Sent successfull with ACK")
        else    
            println("Sent failed")
        end
 
        wait(2000)
 
    wend
fend
 
DoTestRx() 
```