ghi-electronics/duelink-website #209: Portal CP70

Palomino34, 2025-05-06:
Seems to work, show logo on screen by using C/C++ successfully

No further tested as sound, touch, flash...

No script available at this time.

Palomino34, 2025-06-06:
this script below work in DUE

```
Alias(cs=10,pd=18,bit32=32,bit24=24,bit16=16,bit8=8,dladd=_a)

_a = 0x00300000

fn SpiWrRd(d, o)
    dim b1[1]
    dim b2[1]
    
    v = 0
    
    if o = bit8
        b1[0] = d
        SpiWrs(b1,b2)
        v = b2[0]
        #println("Write read 8")

    end
    
    if o = bit16
        b1[0] = (d >> 8) & 0xFF
        SpiWrs(b1,b2)
        v = v | (b2[0] << 8)
        
        b1[0] = d & 0xFF
        SpiWrs(b1,b2)
        v = v | (b2[0] << 0)
        #println("Write read 16")
      
        
    end
    
    if o = bit24 
        b1[0] = (d >> 24) & 0xFF
        SpiWrs(b1,b2)
        v = v | (b2[0] << 24)
        #println(hex(b1[0]))
        
        b1[0] = (d>>16) & 0xFF
        SpiWrs(b1,b2)
        v = v | ((b2[0]) << 16)
        #println(hex(b1[0]))
        
        b1[0] = (d>>8) & 0xFF
        SpiWrs(b1,b2)
        v = v | ((b2[0]) << 8)
        #println(hex(b1[0]))
        
        #println("Write read 24")
        
    
    end
    
    if o = bit32
        b1[0] = (d >> 24) & 0xFF
        SpiWrs(b1,b2)
        v = v | (b2[0] << 24)
        #println(hex(b1[0]))
        
        b1[0] = (d>>16) & 0xFF
        SpiWrs(b1,b2)
        v = v | ((b2[0]) << 16)
        #println(hex(b1[0]))
        
        b1[0] = (d>>8) & 0xFF
        SpiWrs(b1,b2)
        v = v | ((b2[0]) << 8)
        #println(hex(b1[0]))
[... 195 lines cut ...]
Synth(0x00, 0)

# Draw GHI
DLClear()
DLBuild(  [0x02,0x00,0x55,0x00] )
DLBuild(  [0x26,0x00,0x00,0x07] )
DLBuild(  [0x1F,0x00,0x00,0x01] )
DLBuild(  [0x9B,0x86,0xEF,0xC7] )
DLBuild(  [0x9E,0x86,0xEF,0xC8] )
DLBuild(  [0xA1,0xC6,0xEF,0xC9] )
DLBuild(  [0xA5,0x66,0xEF,0xAE] )
DLBuild(  [0x21,0x00,0x00,0x00] )
DLBuild(  [0x04,0xA0,0x16,0x16] )
DLBuild(  [0x0D,0x00,0x01,0x40] )
DLBuild(  [0x1F,0x00,0x00,0x02] )
DLBuild(  [0x98,0x08,0x50,0x00] )
DLBuild(  [0x04,0xFF,0xFF,0x7F] )
DLBuild(  [0x1F,0x00,0x00,0x03] )
DLBuild(  [0x42,0x68,0x03,0xD0] )
DLBuild(  [0x43,0x60,0x06,0x50] )
DLBuild(  [0x21,0x00,0x00,0x00] )
DLBuild(  [0x00,0x00,0x00,0x00] )

DLSwap()
```

Palomino34, 2025-06-06:
It is translated from C/C++ below:

```
uint8_t regGpio;
	int i;

    
    uint8_t val;

	MCU_Init(); // Just init spi pin, CS, DP

    // ------------------------- System settings ------------------------------
	// Set Chip Select OFF
	HAL_ChipSelect(0);

	// Reset the display
	HAL_Delay(20);
	HAL_PowerDown(1);
	HAL_Delay(20);
	HAL_PowerDown(0);
	HAL_Delay(20);

	HAL_HostCmdWrite(0x44, 0x00); // 0x44 = HostCMD_CLKEXT
	HAL_HostCmdWrite(0x61, 0x46);

    HAL_HostCmdWrite(0x68, 0x00); // Reset

	// Set active
	HAL_HostCmdWrite(0, 0x00);
	
	HAL_Delay(500);		// Optional delay can be commented so long as we check the REG_ID and REG_CPURESET

	// Read REG_ID register (0x302000) until reads 0x7C
	while ((val = HAL_MemRead8(EVE_REG_ID)) != 0x7C)//EVE_REG_ID
	{
	}

	// Ensure CPUreset register reads 0 and so FT8xx is ready
	while (HAL_MemRead8(EVE_REG_CPURESET) != 0x00)
	{
	}

    HAL_MemWrite32(EVE_REG_FREQUENCY, 72000000);
    

	// ------------------------- Display settings ------------------------------

	// LCD display parameters
	// Active width of LCD display
	HAL_MemWrite16(EVE_REG_HSIZE,   EVE_DISP_WIDTH);
	// Total number of clocks per line
	HAL_MemWrite16(EVE_REG_HCYCLE,  EVE_DISP_HCYCLE);
	// Start of active line
	HAL_MemWrite16(EVE_REG_HOFFSET, EVE_DISP_HOFFSET);
	// Start of horizontal sync pulse
	HAL_MemWrite16(EVE_REG_HSYNC0,  EVE_DISP_HSYNC0);
	// End of horizontal sync pulse
	HAL_MemWrite16(EVE_REG_HSYNC1,  EVE_DISP_HSYNC1);
	// Active height of LCD display
	HAL_MemWrite16(EVE_REG_VSIZE,   EVE_DISP_HEIGHT);
	// Total number of lines per screen
	HAL_MemWrite16(EVE_REG_VCYCLE,  EVE_DISP_VCYCLE);
	// Start of active screen
	HAL_MemWrite16(EVE_REG_VOFFSET, EVE_DISP_VOFFSET);
	// Start of vertical sync pulse
	HAL_MemWrite16(EVE_REG_VSYNC0,  EVE_DISP_VSYNC0);
	// End of vertical sync pulse
	HAL_MemWrite16(EVE_REG_VSYNC1,  EVE_DISP_VSYNC1);
	// Define RGB output pins
	HAL_MemWrite8(EVE_REG_SWIZZLE,  EVE_DISP_SWIZZLE);
	// Define active edge of PCLK
	HAL_MemWrite8(EVE_REG_PCLK_POL, EVE_DISP_PCLKPOL);
	// Turn on or off CSpread
	HAL_MemWrite8(EVE_REG_CSPREAD,  EVE_DISP_CSPREAD);
	// Turn on or off Dither
	HAL_MemWrite8(EVE_REG_DITHER,  EVE_DISP_DITHER);

	// Write first display list
	HAL_MemWrite32((EVE_RAM_DL + 0), EVE_ENC_CLEAR_COLOR_RGB(0,0,0));
	HAL_MemWrite32((EVE_RAM_DL + 4), EVE_ENC_CLEAR(1,1,1));
	HAL_MemWrite32((EVE_RAM_DL + 8), EVE_ENC_DISPLAY());
	HAL_MemWrite8(EVE_REG_DLSWAP, EVE_DLSWAP_FRAME);

	// Read the  GPIO register for a read/modify/write operation
	regGpio = HAL_MemRead8(EVE_REG_GPIO);
	// set bit 7 of  GPIO register (DISP) - others are inputs
	regGpio = regGpio | 0x80;
	// Enable the DISP signal to the LCD panel
	HAL_MemWrite8(EVE_REG_GPIO, regGpio);
    HAL_MemWrite8(EVE_REG_PCLK, EVE_DISP_PCLK);
	HAL_MemWrite8(EVE_REG_PWM_DUTY, 127);
        

	// ---------------------- Touch and Audio settings -------------------------
	// Eliminate any false touches
	HAL_MemWrite16(EVE_REG_TOUCH_RZTHRESH, 1200);

	// turn recorded audio volume down
	HAL_MemWrite8(EVE_REG_VOL_PB, EVE_VOL_ZERO);
	// turn synthesizer volume down
	HAL_MemWrite8(EVE_REG_VOL_SOUND, EVE_VOL_ZERO);
	// set synthesizer to mute
	HAL_MemWrite16(EVE_REG_SOUND, 0x6000);
    
    // GHI 
    EVE_TestPlaySound(); // test Sound
    
    HAL_MemWrite32(EVE_RAM_DL + 0, EVE_ENC_CLEAR_COLOR_RGB(0, 85, 0));
    HAL_MemWrite32(EVE_RAM_DL + 4, EVE_ENC_CLEAR(1, 1, 1)); // clear screen
    HAL_MemWrite32(EVE_RAM_DL + 8, EVE_ENC_BEGIN(EVE_BEGIN_BITMAPS)); // start drawing bitmaps
    HAL_MemWrite32(EVE_RAM_DL + 12, EVE_ENC_VERTEX2II(220, 110, 31, 'G')); // ASCII T in font 31
    HAL_MemWrite32(EVE_RAM_DL + 16,EVE_ENC_VERTEX2II(244, 110, 31, 'H')); // ASCII E in font 31
    HAL_MemWrite32(EVE_RAM_DL + 20, EVE_ENC_VERTEX2II(270, 110, 31,'I')); // ASCII X in font 31
    HAL_MemWrite32(EVE_RAM_DL + 24, EVE_ENC_VERTEX2II(299, 110, 31, '.')); // ASCII T in font 31
    HAL_MemWrite32(EVE_RAM_DL + 28, EVE_ENC_END());
    HAL_MemWrite32(EVE_RAM_DL + 32, EVE_ENC_COLOR_RGB(160, 22, 22)); // change colour to red
    HAL_MemWrite32(EVE_RAM_DL + 36, EVE_ENC_POINT_SIZE(320)); // set point size to 20 pixels in radius
    HAL_MemWrite32(EVE_RAM_DL + 40, EVE_ENC_BEGIN(EVE_BEGIN_POINTS)); // start drawing points
    HAL_MemWrite32(EVE_RAM_DL + 44, EVE_ENC_VERTEX2II(192, 133, 0, 0)); // red point
         
    HAL_MemWrite32(EVE_RAM_DL + 48, EVE_ENC_COLOR_RGB(255, 255, 127));
    HAL_MemWrite32(EVE_RAM_DL + 52, EVE_ENC_BEGIN(EVE_BEGIN_LINES));
    HAL_MemWrite32(EVE_RAM_DL + 56, EVE_ENC_VERTEX2F(1232, 976));
    HAL_MemWrite32(EVE_RAM_DL + 60, EVE_ENC_VERTEX2F(1728, 1616));

    HAL_MemWrite32(EVE_RAM_DL + 64, EVE_ENC_END());
    HAL_MemWrite32(EVE_RAM_DL + 68, EVE_ENC_DISPLAY()); // display the image
       
    HAL_MemWrite8(EVE_REG_DLSWAP, EVE_DLSWAP_FRAME);

   
    while(1);
}

```

taylorza, 2025-06-15:
I have a proposed change to the driver, so the code can look a little more like the C code without needing to create lots of functions. We might need to implement a few C helpers to improve performance.

```
Alias(cs=10,pd=18,bit32=32,bit24=24,bit16=16,bit8=8,dladd=_a)

_a = 0x00300000
dim b3[4]

fn SpiWrRd(d, o)
    dim b1[1]
    dim b2[1]
    
    v = 0
    if o = bit8
        b1[0] = d
        SpiWrs(b1,b2)
        v = b2[0]
        #println("Write read 8")
    end
    
    if o = bit16
        b1[0] = (d >> 8) & 0xFF
        SpiWrs(b1,b2)
        v = v | (b2[0] << 8)
        b1[0] = d & 0xFF
        SpiWrs(b1,b2)
        v = v | (b2[0] << 0)
        #println("Write read 16")
    end
    
    if o = bit24 
        b1[0] = (d >> 24) & 0xFF
        SpiWrs(b1,b2)
        v = v | (b2[0] << 24)
        #println(hex(b1[0]))
        b1[0] = (d>>16) & 0xFF
        SpiWrs(b1,b2)
        v = v | ((b2[0]) << 16)
        #println(hex(b1[0]))
        b1[0] = (d>>8) & 0xFF
        SpiWrs(b1,b2)
        v = v | ((b2[0]) << 8)
        #println(hex(b1[0]))
        #println("Write read 24")
    end
    
    if o = bit32
        b1[0] = (d >> 24) & 0xFF
        SpiWrs(b1,b2)
        v = v | (b2[0] << 24)
        #println(hex(b1[0]))
        b1[0] = (d>>16) & 0xFF
        SpiWrs(b1,b2)
        v = v | ((b2[0]) << 16)
        #println(hex(b1[0]))
        b1[0] = (d>>8) & 0xFF
        SpiWrs(b1,b2)
        v = v | ((b2[0]) << 8)
        #println(hex(b1[0]))
        b1[0] = (d>>0) & 0xFF
        SpiWrs(b1,b2)
        v = v | ((b2[0]) << 0)
        #println(hex(b1[0]))
        #println("Write read 32")
    end 
    return v
fend

fn SetAdd(a,r)
    if (r=1) # read
[... 220 lines cut ...]
  DLCmd(C_VERTEX2F, {0,y})
  DLCmd(C_VERTEX2F, {6400,y})
next
DLCmd(C_END,{})

DLCmd(C_COLOR,{0,0,255})
DLCmd(C_BEGIN,{BITMAPS})
DLCmd(C_VERTEX2II, {150,10,31,'G'})
DLCmd(C_VERTEX2II, {185,10,31,'H'})
DLCmd(C_VERTEX2II, {220,10,31,'I'})
DLCmd(C_END,{})

DLCmd(C_COLOR,{255,0,255})
DLCmd(C_BEGIN,{POINTS})
DLCmd(C_POINT_SIZE,{20<<4})
DLCmd(C_VERTEX2F, {192<<4,1350})
DLCmd(C_POINT_SIZE,{18<<4})
DLCmd(C_COLOR,{0,0,0})
DLCmd(C_VERTEX2F, {192<<4,1350})
DLCmd(C_END,{})

DLCmd(C_DISPLAY,{})

DLSwap()
```

closed 2025-08-15 by gus-ghielec